Mechanical-Advantage/AdvantageScope #1: Advantage Scope Arch Linux

Haadi-Khan, 2022-02-20:
Hi,

I wanted to try out this utility for my team, and I use Endeavour OS on my laptop primarily (Windows bogs it down a ton). I wanted to try out the software using my regular installation instead of using my Windows install. 

Would it be possible for some instructions on how to run this on my current installation (since there's no prebuilt binary)? If it's too much of a hassle however, I'm fine just using Windows.

Regardless, thanks for creating this utility, and I'm excited to see if we can use it to benefit our team.

jonahb55, 2022-02-20:
I just released [Advantage Scope 1.16.1](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v1.16.1), which includes a `.pacman` file in the Linux zip. I don't have a machine to test it on, so let me know if it works properly. For now, this means that the Linux zip is rather large as it includes `.deb`, `.rpm`, and `.pacman` files; these could probably be split up in the future.

You should also be able to run Advantage Scope from the command line if necessary. Just install Node.js, clone the repository, and run the  following commands:
```
npm install
npm start
```

Haadi-Khan, 2022-02-21:
Sorry, didn't get a chance to try it out till now. Just tried it, and it worked. Excited to try this out tomorrow.

If anyone stumbles across this at a later point, you can install via a ".pacman" file by doing `pacman -U "File Name"`.

closed 2022-02-21 by Haadi-Khan