Mechanical-Advantage/AdvantageScope #8: Connecting to a live log does not include previous logs

jonahsnider, 2022-03-20:
Reproduction:
1. Start a robot program
2. Connect to the live log
3. Observe that the first timestamp shown is not `0`

jonahb55, 2022-03-22:
This isn't a bug so much as a practical limitation. When a new client connects, it begins receiving the stream of data from each subsequent cycle. While theoretically possible, I think there are good reasons not to send older log data:

* Right now, AdvantageKit doesn't keep old log data after saving it to an RLOG file (if it's even configured to do that, which it might not be in a simulator). It would likely need to store this data in memory, which is clearly not optimal for performance (especially during long sessions).
* If the data was available, it would take *at least* a few seconds to send the data over the network and decode it up front. This seems like it would be quite inconvenient when starting a debug session.
* Overall, the optimal way to accomplish this would probably be to stream sections of log data over the network when requested. Setting that up would be a large undertaking though, especially because it's not really compatible with the RLOG format right now (which depends on being read sequentially).

If something odd happens when we're testing the robot without Advantage Scope running, we usually just pull the last log file and look at that. This is possible even when the file is being actively updated by the robot program. While perhaps not as convenient as connecting to the server, I'd say this is a reasonable workaround given the complexity of the alternative.

Also see my comment on #7 about why that appeared to fix the issue.

jonahsnider, 2022-03-22:
Those are all valid points. Perhaps in the future if the RLOG format changes we can reevaluate the pros & cons of implementing this feature.

closed 2022-03-22 by jonahsnider