Mechanical-Advantage/AdvantageScope #12: wpiLog support

m10653, 2022-04-19:
Are there plans on supporting importing log files from wpilib's logging system. At least in the capability of graphing data stored within them?

jonahb55, 2022-04-19:
Better integration with WPILib's built-in logging (including wpilogs) is something that we're planning to evaluate during the offseason, which could include switching AdvantageKit to use wpilogs instead of our custom RLOG format. We'll also take into account that until [Sapphire](https://github.com/wpilibsuite/allwpilib/projects/14) is more complete, there would be some utility to teams in supporting wpilogs with Advantage Scope.

We can get back to you once we have time to properly evaluate all of these possibilities.

jwbonner, 2022-11-13:
This has been added in the 2023 beta release of AdvantageScope.

https://www.chiefdelphi.com/t/advantagescope-2023-beta-live-visualization-log-analysis-with-wpilog-nt4/417424?u=jonahb55

closed 2022-11-13 by jwbonner