Mechanical-Advantage/AdvantageScope #16: Driver controller visualization

plusparth, 2022-10-04:
It would be really useful to have some type of visualization for which buttons are pressed, how far the joystick/triggers are pushed, etc in AdvantageScope to allow for quickly debugging to see if some observed robot issue was caused by driver error or by some software bug.

The data should already be logged by default with AdvantageKit but it requires some effort to check which buttons and axes correspond to which physical controls. Ideally this could be visualized with an image of a controller highlighting pressed buttons and showing where the joysticks are, etc, for a couple of commonly used controllers.

jonahb55, 2022-10-04:
By complete coincidence, I'm already in the midst of implementing this exact feature. We had a problem at our last offseason that was a perfect use case (we spent a while trying to track down a software problem only to find the driver pushed the wrong button). I'll post a screenshot once I get it working.

jonahb55, 2022-10-10:
This feature is working nicely and will definitely be included in the first 2023 beta release. I included a collection of commonly used controllers:

[image]

You can also do a generic grid view or add custom joysticks (like 6328's override console in this example).

[image]

closed 2022-10-10 by jonahb55