Mechanical-Advantage/AdvantageScope #22: 3d viewer robot rotation does not align with simulation

gerth2, 2022-12-17:
https://user-images.githubusercontent.com/4583662/208266877-2a65a343-95fc-4d67-8059-9ce3a7ede597.mp4

Looks like a potential units issue? Radians vs. degrees? the pose in NT is broadcasting as degrees, but in AdvantageScope it might be interpreting as radians?

Field2D broadcasts in Degrees for sure, and that's what the sim GUI seems to like:

[image]

jwbonner, 2022-12-17:
Yes, this is definitely a units issue. Right now the rotation units are configuration for the odometry tab, but the 3D field always uses radians. I'm hoping to release the next beta sometime this week, which will make this configurable along with a bunch of other improvements:

[image]

jwbonner, 2022-12-22:
The rotation units are now configurable in [v2.0.0-beta-2](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v2.0.0-beta-2). Under "Config" > "Units" select "degrees."

closed 2022-12-22 by jwbonner