Mechanical-Advantage/AdvantageScope #23: Feature suggestion: select all when downloading robot logs

We-Gold, 2022-12-20:
It's great that advantagescope has the ability to download log files from the robot, however we often find ourselves downloading many at a time, and it can take unnecessary time to select all of them. I find myself reaching for the datalogtool for downloading for this reason.

It would be great to see this feature, and maybe even more information about the downloading process (estimated time remaining, etc.)

jwbonner, 2022-12-20:
If you weren't aware, you can select multiple files quickly using shift-click (the same as most file browsers):

[image]

Showing the download progress and other info should be a relatively easy addition. I'll see what I can do.

We-Gold, 2022-12-20:
I saw the shift click, but my brain went to ctrl/cmd + a immediately. Shift clicking is probably enough, but ctrl a could be nice too.

jwbonner, 2022-12-22:
These improvements are now added in [v2.0.0-beta-2](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v2.0.0-beta-2). I implemented the ctrl/cmd + A shortcut, plus some other quality of life improvements:

[image]

closed 2022-12-22 by jwbonner