Mechanical-Advantage/AdvantageScope #28: Odometry Back Front Arrows Origins Backwards

tom131313, 2022-12-30:
Odometry tab the depiction of the Robot with the arrow in box pointing forward matches the PathPlanner assumption of the front of the robot. Selecting Arrow (Back, Center, Front) would seem to be backwards of that assumed orientation. It's hard to guess what was meant by the arrows since the (Front) is more left of the (Back) arrow if pointing to the (Right).

jwbonner, 2022-12-30:
The "Front"/"Center"/"Back" options for the arrows indicate whether the provided pose is aligned to the tip, center, or back of the arrow - it's just shifting it forward or back.  Which of those makes the most sense may be different depending on the use case (e.g. a robot pose could be aligned to the center of the arrow while a vision target on the edge of the field could be aligned to the back). I understand why the terminology may be confusing though, given that the robot also has a "front" and "back." Maybe "Tip"/"Centered"/"Base" would be an improvement?

tom131313, 2022-12-30:
It took me a couple of minutes but I get what you are saying about the pose/arrow alignment. I don't see why not to use the robot rectangular shape like the Robot option does and have a "target" circle on that robot to indicate the location of the pose. 

Indeed, the Robot option has some assumed pose location and that was the first and immediate question I had was where was the robot drawn relative to the pose?

Eliminating that vague Robot icon for three Robots that include a target circle at the back, center, or front of the robot rectangle would clarify the situation. Your words as they are now - back, center, front - then make sense. [Robot (pose at back], Robot (pose at center), Robot (pose at front)]

jwbonner, 2022-12-30:
The robot is centered on the pose because that's the standard convention used by WPILib, PathPlanner, other dashboards, etc. I suspect that adding options for the robot pose is overly confusing because it doesn't follow the conventions of the rest of the ecosystem.

I think the arrows are still useful for poses that don't indicate the robot's location, like vision targets. You can always add multiple robot/ghost objects, but the arrows are more flexible for advanced use cases. I experimented with a minor redesign for the arrows that may make the difference between them more clear. Do you think this an improvement?

[image]
[image]
[image]

tom131313, 2022-12-31:
Yes, thank you, much improved. It might not be easily obvious that the cross bar is the point of the pose but with a little documentation describing that, it's easy to use. The current way has no cue as to where the pose is especially since there is no path drawn on the screen; your suggested way completely solves that problem.

It is more obvious the the robot pose is centered on the robot since there is the path drawn. You might consider putting the cross bar on the center of the robot, too, for consistency and another quick cue but it's not essential for use.

commit b5573d5e (2023-01-02): Add crossbars to odometry arrows (closes #28)
closed 2023-01-02 by jwbonner