Mechanical-Advantage/AdvantageScope #29: Odometry Tab Extra Drawn From End to Start

tom131313, 2022-12-30:
The path is correctly drawn from the start at the left of the image and up and down to the end but an "extra" line is drawn from the end to the start no matter how the time line completes - either by completing the path or by pressing the time line Start/Pause icon.

Then as the time line continues to play somewhat past the path end time the robot resets to its start position and stays there until the time line runs to the end and then the robot end position is redisplayed.
[image]

jwbonner, 2022-12-30:
Could you send the log file that exhibits this issue? If you're connected over NetworkTables you can export the data as a WPILOG under "File" > "Export Data..." > "Format" > "WPILOG". You may need to put it in a zip file to upload it to GitHub.

tom131313, 2022-12-30:
I looked at the Line Graph of the data and I think I see what is happening. The odometry and line graph are time series. I provide about 5.23 seconds of data and the "page width" is about 58 seconds. AdvantageScope charts my 5.23 seconds, holds the final data values for about 6 more seconds then fills the remainder of the 58 seconds with the initial data values. I'm not sure that's wrong but it was surprising and hard to figure out by just looking at the odometry and when coupled with my other problem.

For some reason the odometry reverted to the initial pose when pausing the time line. I think I fixed that problem by adding a `logEntry.finish( (long)(1_000_000.*trajectory.getEndState().timeSeconds) );`
At least I can't reproduce the problem today with this line of code inserted.

I like the idea of AdvantageScope and have been looking for such a tool for a long time and it works for the WPILib DataLog which has a huge restriction with its array element delimiter (";"). I wish there was more documentation and examples as I struggled to get log entries to its satisfaction (the log timestamps were problematic and I still don't understand the various types of poses). The Field2d reference is baffling and I suspect it is refering to "real-time" charting that doesn't work for me since AdvantageScope won't connect to my roboRIO. I reported a problem and I am hopeful it's just the difference between 2022 and 2023 NT4 and I'll be able to use NT4 next week.

After many failures this is what I got to work for swerve pose odometry:
```
PathPlannerTrajectory trajectory = PathPlanner.loadPath("PP1", new PathConstraints(4, 3));
DoubleArrayLogEntry dale;
DataLogManager.start();
DataLog log = DataLogManager.getLog();

dale = new DoubleArrayLogEntry(log, "Trajectory", "Swerve Pose");

for(int i = 0; i < trajectory.getStates().size(); i++)
   {    
       var state = trajectory.getState(i);
       dale.append(new double[] {
           state.poseMeters.getTranslation().getX(),
           state.poseMeters.getTranslation().getY(),
           state.holonomicRotation.getRadians() // pose doesn't have swerve rotation, only direction of travel
           },
           (long)(1_000_000.*state.timeSeconds));
   }
dale.finish( (long)(1_000_000.*trajectory.getEndState().timeSeconds) );
```
[image]

jwbonner, 2022-12-30:
I'm glad you figured out the issue. I believe that WPILib's recommended method for data logging is to publish to NT and have the `DataLogManager` pull that data automatically. Field2d is a wrapper for publishing the robot pose (and optionally other objects like trajectories) to NT, so it would be included in the log in that case. I haven't been able to recreate the networking issues you've seen, but hopefully that's resolved with the kickoff release of WPILib. If you have suggestions related to documentation for WPILib's data logs or NT telemetry, I would recommend opening an issue on [frc-docs](https://github.com/wpilibsuite/frc-docs/issues).

closed 2025-03-18 by jwbonner