Mechanical-Advantage/AdvantageScope #39: Improve value Field Accessibility with tooltips

m10653, 2023-01-28:
I added the following feature that displays a warning triangle on value fields that are missing their respective keys. On hover a message is displayed explaining in detail what the warning is about. Below is a rough example of what it looks like.
[image]

I thought that it would be nice to make the type selection for displaying data a bit smarter as well as providing feedback when things are not displaying due to an incompatible type. IE display an warning that data is not being displayed when someone selects pose3d and are trying to plot AprilTag ID's. Furthermore try to improve how intelligent the default type selector is or provide potential solutions to potently compatible types. IE warning "This data does not appear to be a Pose3d. Possibly a "Trajectory, AprilTagID, etc"

Before I spend a bunch of time implementing it would this be feature you could see being added?

jwbonner, 2023-01-28:
I think having more descriptive errors for missing/invalid fields is a good idea. Just a couple of things to consider:

- The behavior of crossing out missing fields is used across many different tabs (line graph, table, console, stats, odometry, 3D field, swerve, mechanism, and points). If you add more descriptive icons/errors, there should be consistent behavior across all of the places where fields can be added by the user (and therefore could be missing).
- It's hard to tell from the screenshot you posted how exactly the tooltip is displayed. For consistency with the other places tooltips/popups work in AdvantageScope, I would suggest either using the HTML [title attribute](https://www.w3schools.com/TAGS/att_title.asp) (which AdvantageScope uses in some places when field keys are cut off) or doing an alert popup on the window (like [this](https://github.com/Mechanical-Advantage/AdvantageScope/blob/63bfd91ae912600ac9906055048956e56940abcd/src/hub/tabControllers/StatisticsController.ts#L62)).
- Feel free to look at identifying object types, though that's definitely a trickier problem. For example with poses you could check if the length of the array is a multiply of 7 (Pose3d) or 3 (Pose2d) but it's always going to be ambiguous in some circumstances (what if the array has 21 items?) AprilTag IDs are probably simpler since they should all be integers from 0 to 29. I'm not opposed to autodetecting object types as long as the implementation avoids any false positives.

m10653, 2023-01-28:
- So the tooltip is done via CSS. On hover over(Screenshot hides the cursor) the triangle it displays a hidden span that is part of the warning triangle. There are some issues with this approach as there is some Z-index issues with the table col to the side as well on some screen resolutions the message being cut off. I do not think the popup window would be a good fit but ill look into the title attrb.
- So the behavior for that more descriptive error (Just for the missing element) is within the `TimelineVizController` so it should apply to any windows that extend that class. For fancy type checking it is likely the code would have to be added for a per tab basis. But I have yet to come up with a clean idea on how to implement that.
- For auto detecting object types. The idea for that was to improve on the current behavior of the dropdowns. Currently the first field you put down it selects Robot-> then Ghost -> etc regardless of the time. The improvement would be taking into account the size of the data and skipping the options that would not make any sense. IE you drop a April tag id into a empty tab.  Instead of directly picking Robot it would be able to infer a better option like the April tag ids. There is also the option for smarter detection for the April tag ids because they should in theory be equal to the number of April tag 3dpose objects in its sister field.  In this situation a false positive would not be a huge deal because that is already the current behavior to just follow a predictable pattern.
- The big quality of life improvement at minimum I wanted to add was some sort of indication that parsing the data has failed (based on type) and is not being displayed/used in the window.