Mechanical-Advantage/AdvantageScope #42: High update rate causes disconnects from RoboRIO when there are many topics. Need a way to set lower update rate.

JohnGilb, 2023-02-04:
**Summary**:
When you have a lot of network table entries and attempt to connect to a live robot using AdvantageScope, the Network Tables server disconnects you.

**Why does this happen?**
According to the WPILib contributor Peter (excerpts of quotes from unofficial FIRST discord):

_"it's taking a significant amount of time to send the list of topics over.  meanwhile, it's starting to send value updates every 5 ms.  it checks to see if the transmit buffer is full 10 times (this is a multiple of the update time, not a fixed timer), if that happens it thinks the transmit is stalled and closes the connection."_

_"advantagescope has ... aggressive subscriber options.  .. periodic=0 (which is forced to the minimum of 5 ms by the server), send all = true"_

**What could be done?**
If there was a way to configure the requested update rate on the AdvantageScope side (the NT default is 100ms), users with many network table entries could still use AdvantageScope live with reduced fidelity. This would be fine, since we can always recover the high-fidelity logs from the robot after the fact.

jwbonner, 2023-02-05:
Could you check if b44b33a fixes your issue? You can download the dev build from GitHub Actions [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/runs/4094561780). This subscribes with a 250ms period at first (to work around the NT server bug where the transmit buffer fills up while sending the list of topics), then it lowers to a 20ms period after one second. Using 20ms instead of 5ms should reduce the server's overly aggressive disconnect behavior.

JohnGilb, 2023-02-05:
Thanks, @jwbonner. I will have the opportunity to test later today.

jwbonner, 2023-02-05:
No problem. It looks like Peter fixed some of the issues on the server side too in WPILib 2023.3.1. I'll put out an AdvantageScope release later today.

closed 2023-02-05 by jwbonner