Mechanical-Advantage/AdvantageScope #49: Layout glitch: April tag Locations off when on Red Alliance

randomstring, 2023-02-16:
[image]

When on the red alliance and using the WPILib to change the layout origin, the april tags are off by a bit. Front AprilTag is what AdvantageScope is showing, the smiley face AprilTag is the 3dPose of the transformed apriltag computed by WPIlib `AprilTagFieldLayout.setOrigin()`

```        
  layout.setOrigin(OriginPosition.kRedAllianceWallRightSide);
  Logger.getInstance().recordOutput("Vision/3d_AprilTags/1", layout.getTagPose(1).get());
```

The reported AprilTag 3dPose is [0.9456419999999985,7.157973999999999,0.462788,1,0,0,6.123233995736766e-17]
(functionally equivalent to  [0.945642,7.157974,0.462788,1,0,0,0])

I'm wondering where the error is coming from? Is it in the AdvantageScope 3d map?

randomstring, 2023-02-16:
So the field layout config file lists the field dimensions as:
```json
      "length": 16.4592, 
      "width": 8.2296
```

The game manual says the length field is 54 ft. 3 1⁄4 in or 651.25" or 16.54175 meters. A difference of 3.25"
field width is 315.5" or 8.0137 meters. off by 8.5". Maybe we missed an update somewhere.

jwbonner, 2023-02-16:
Where are you getting that config JSON? I see 16.54175 meters in WPILib and it hasn't been updated since January 8th when the PR was merged: https://github.com/wpilibsuite/allwpilib/blob/main/apriltag/src/main/native/resources/edu/wpi/first/apriltag/2023-chargedup.json#L149

AdvantageScope uses 651.25 inches: https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/frcData/Field3d_2023.json#L7

jwbonner, 2023-02-16:
Here's what I see with a red alliance origin on WPILib 2023.3.2 (loaded with `AprilTagFields.k2023ChargedUp.loadAprilTagLayoutField()`) and AdvantageScope 2.2.1:

[image]

closed 2023-02-16 by jwbonner

randomstring, 2023-02-16:
Thanks. We had an old (or just wrong?) version of the field setup.

for those who come after, load the field data this way and not from your own file.

```
AprilTagFields.k2023ChargedUp.loadAprilTagFieldLayout();
```