Mechanical-Advantage/AdvantageScope #51: DS log current issue

TylerSeiford, 2023-02-19:
When viewing a log from our robot, PDH channel 18 shows 0 current draw.  Here's a comparison of the default viewer and advantage scope.

[image]

[image]

TylerSeiford, 2023-02-19:
Channels 15, 16 & 17 (the rest of our drive train) all show correctly.

rzblue, 2023-03-10:
Just wanted to chime in here- I don't think AdvantageScope is reading the DS log currents correctly (at least not with the PDH).

 Here are two screenshots from the same log, one in DS Log viewer, one in advantagescope. I've isolated channel 19 as that seems to be the one that most noticable has this issue in our logs. The images are zoomed in on the same 7 second period, the first 7 seconds of auto at the start of one of our matches.
 
 In advantagescope, the current draw value is always a discrete value - 0, 32, 64, or 96. The DS Log viewer shows the accurate (and more believable) reading.
 
 
[image]
[image]

jwbonner, 2023-10-15:
This should now be fixed (d6b0569). As far as I can tell, this was only an issue with the PDH and not the PDP. For comparison, here's what the two logs posted here look like after the fix:

@TylerSeiford

[image]

@rzblue

[image]

closed 2023-10-15 by jwbonner