Mechanical-Advantage/AdvantageScope #53: Make wpilog and live data have the same prefix

scarmain, 2023-02-23:
Currently, if you setup a template to use WpiLib log data, keys are prefixed with a "NT:" ("NT:/FMSInfo/MatchNumber").  But if you connect to the robot via IP or a virtual robot with simulation using NetworkTables, the keys are root based ("/FMSInfo/MatchNumber").

Is there a way you can make this universal so that we can share a dashboard between looking at live robot data and logs and not have to change all the key names?

TheTripleV, 2023-03-22:
I'd rather add the NT prefix to the live view.

jwbonner, 2023-03-28:
I put together a potential fix for this along the lines of @TheTripleV's suggestion. When in the default "NetworkTables 4" mode, the "NT:" prefix is added to all of the fields. I also set up the sidebar to identify when only one table is available in the data, then use it as the root (so it appears the same as before). This approach should mean that the user never needs to think about this and it will "just work."

@scarmain @TheTripleV @17acres Does this seem like a good solution to the problem? You can download the CI artifacts [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/runs/4546753141) if you want to try it out (once the builds finish).

17acres, 2023-03-31:
This seems like a better solution, works for me.

closed 2023-08-28 by jwbonner