Mechanical-Advantage/AdvantageScope #64: Moving window for graph page

We-Gold, 2023-03-28:
While graphing data, we often don't need to see the entire history of the data coming through. We can of course refresh and reset the window, but that is inconvenient and doesn't work exactly the way we'd like. It would be preferable to have the option to view data in a moving window rather than seeing all historical data.

jwbonner, 2023-03-29:
You can zoom in to the line graph by scrolling vertically, then pan left and right by scrolling horizontally or by clicking and dragging. The view will auto-scroll by default when connected to a live source, but that can be toggled by clicking the red arrow in the title bar or scrolling to the left. There are more details in the [line graph](https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/tabs/LINE-GRAPH.md) and [live sources](https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/OPEN-LIVE.md#viewing-live-data) docs.

jwbonner, 2023-09-12:
Closing this issue since I believe the issue was resolved.

closed 2023-09-12 by jwbonner