Mechanical-Advantage/AdvantageScope #65: [Request] Organization Solutions

MidgleyM, 2023-04-02:
Its my teams first year using Advantage Kit, and its great, we haven't had a ton of issues. but as the person that is responsible for my teams logging, and going through that data and making it useful for us, its pretty annoying and difficult to look through everything we have logged just to find the one thing that I want, and it is even worse for those that don't know what we have logged. I know that things are alphabetically sorted, but it would be much more compact and readable if we could put the values in a folder, like what is done with the rail voltages in system stats. It might be possible to do it with sorting through the values name, and if it has a dash in it, then what comes before that is the folder to put it in, for example, pose/limelightPose would be put in a folder named pose. I don't know if this is doable, or if there is already a solution that I just don't know about, but I would appreciate any solution, even if its on Advantage Kit, or Advantage Scope.

jwbonner, 2023-04-02:
What you’re describing is already supported by logging a value with forward slashes in the key, so “Drive/Module0/Position” would be in nested folders for “Drive” and “Module”. It sounds like you already had that idea, so are you having trouble making it work?

jwbonner, 2023-09-12:
Closing this issue since I believe the issue was resolved.

closed 2023-09-12 by jwbonner