Mechanical-Advantage/AdvantageScope #66: Add support for Zebra MotionWorks™ data

jamesdooley4, 2023-04-04:
We've started using the Zebra MotionWorks™ data from our matches in the PNW district to compare their robot position with what our pose estimation system produces using AprilTag + odometry data and debug cases where the pose estimator goes awry. It'd be useful to visualize the Zebra data alongside other fields, though some time of time offset is needed since the Zebra data timestamps don't quite match with the WPILib logs timestamps (off by a few seconds or so). The Zebra data is also in feet instead of meters or inches - and appears to always be offset from the red side of the field.

jamesdooley4, 2023-04-04:
I've started on one way of doing this in https://github.com/jamesdooley4/AdvantageScope/pull/2

jwbonner, 2023-09-13:
Implemented by 9ac4793b5d1f76ff9002c0546fe482c7df50f5c9 and bbb2203c943fe67617ff2a4bf4633b4b33272d9f. This works by downloading data directly from TBA. It also uses the new 2024 type annotations to correctly handle unit/coordinate system conversions, type options, and interpolation for the odometry and 3D fields (while leaving the raw data available for other tabs).