Mechanical-Advantage/AdvantageScope #70: Drivestation Support & Merging In Control in.

Autumn-Ou, 2023-04-13:
More of a question than an issue, I worked with a few other students to create [In Control](https://github.com/NE-Robotics/In-Control) which is essentially just AdvantageScope with a drive station, a scoring node picker, configurable network tables for bandwidth, & ray casts in 3d field for click to go driving. I was wondering if I should turn this into a fork and PR the features in or if this should stay as a separate project.

jwbonner, 2023-05-01:
Sorry for the late reply. If there are specific features that you feel would benefit AdvantageScope, feel free to PR them individually and we can discuss them in more detail. In the case of NT4 customizability, I think we've addressed the bandwidth concerns with the new [default NT4 mode](https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/OPEN-LIVE.md#live-mode).

Keep in mind that AdvantageScope is designed as a diagnostics tool instead of a driver dashboard, and in general we'd like the feature set to reflect that. You could also take a look at [FRC Web Components](https://github.com/frc-web-components/frc-web-components/) if any of the features you've implemented would be a good fit for that project.

jwbonner, 2024-02-09:
I think this question was answered. If there are features you would like to PR to AdvantageScope those can be discussed separately.

closed 2024-02-09 by jwbonner