Mechanical-Advantage/AdvantageScope #76: Logged Property not Displayed on Line Graph

gcschmit, 2023-04-27:
On one of our Windows 10 laptops, when we are connected to the robot and drag a property onto a line graph, the data is only displayed from that point forward. This is inconvenient as we usual want to see what was just logged in the past. This doesn't occur on some other laptops. This also doesn't occur when we load a saved log file from disk.

I have attempted to remove all remnants of AdvantageScope and reinstall the latest version, but this behavior persists.

If additional information would be helpful, please let me know.

jwbonner, 2023-04-27:
Starting with v2.3.0 AdvantageScope uses a [low bandwidth](https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/OPEN-LIVE.md#live-mode) mode for NT by default. This means that it only subscribes to fields that are actively being viewed. You can switch to the older "logging" mode in the preferences.

I assume that your other laptops are running older versions of AdvantageScope?

gcschmit, 2023-04-27:
Yes, that's exactly it. The other laptops were still on 2.2.2. Thank you!

closed 2023-04-27 by gcschmit