Mechanical-Advantage/AdvantageScope #77: Blurry GUI when on Linux Wayland

viggy96, 2023-07-16:
When using the Wayland display server on Linux, the interface is blurry. Workaround is to start the application with the flags `/opt/AdvantageScope/advantagescope --ozone-platform-hint=auto --enable-features=WaylandWindowDecorations`

Newer versions of Electron don't need these flags, as this is default behavior in those versions. Also, the flags are strangely shown in the titlebar, and I get an error message regarding failing to read a log file when starting with these flags
[image]

jwbonner, 2023-09-17:
The issue with the title bar and error message is now fixed, it was related to how AdvantageScope parses CLI args.

I was able to recreate the issue on the latest version of Electron when fractional window scaling is enabled. With the arguments you included, the app fails to launch at all. Based on the reports of others in https://github.com/electron/electron/issues/33583, this issue seems to be wildly inconsistent across different machines and versions of Chromium. There doesn't seem to be a clear fix until Electron/Chromium/Wayland properly addresses it.

Let me know if you've figured out anything else that might be helpful. The current dev build of AdvantageScope uses Electron 26.2.1 (Chromium 116.0.5845.188). You can download it [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/runs/6210905097) if you want to test it out.

viggy96, 2023-09-18:
Does your CI not publish an Arch package anymore?

jwbonner, 2023-09-18:
I'm actively making changes to the build system for 2024, so the set of formats may change in the future. For now, there are AppImage and Snap builds that should work on any distribution (or you always have the option of building it yourself in any format).

viggy96, 2023-11-07:
I can start AdvantageScope using 
`org.littletonrobotics.advantagescope --ozone-platform-hint=auto` and things work just fine.
I'm using the flatpak.

jwbonner, 2023-11-07:
Nothing has changed for me, but I'll assume the failure to start could be an issue with my desktop environment. I put together a version that automatically inserts the command line flag at startup. It would be helpful if you could try [the build](https://github.com/Mechanical-Advantage/AdvantageScope/actions/runs/6780597491) and see if it fixes the issue (you'll need to use an AppImage build but it behaves the same from my testing). The current Electron version is v26.2.1, which is well after at least some of the changes that should have fixed this by default.

jwbonner, 2026-01-07:
AdvantageScope now uses Electron 39, which auto-detects Wayland by default.