Mechanical-Advantage/AdvantageScope #84: Problem adding custom robot model

mvog2501, 2023-08-28:
There seems to be a problem with importing custom robot models. I started by adding my robot model, formatted as was described in the online documentation, to the folder that "help"/"Show FRC Data Folder" sent me to. After restarting AdvantageScope and my PC (Windows), the robot model did not show up.

Thinking that the issue could have easily been on my part, I downloaded and unzipped the [Robot_BananaSplitV1](https://github.com/Mechanical-Advantage/AdvantageScopeAssets/releases/tag/frc-6328-assets-v1) model and added it to the frcData folder as well. After restarting AdvantageScope again, neither robot showed up in the 3D field robot dropdown.

Looking through past issues, I saw [one ](https://github.com/Mechanical-Advantage/AdvantageScope/issues/35) from earlier this year. Because this seemed to work on January 20th, I backdated my version of AdvantageScope to [v2.2.0]([v2.2.0](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v2.2.0)) and had the same issue.

Looking through the code, I saw that the "frcData" folder was under a constant called LEGACY_ASSETS, so I created a new folder within AppData/Roaming/AdvantageScope called "userAssets" to see if it would fix the issue. It did not. I believe that I tested adding this folder in the most recent release as well, to no avail.

Am I missing something?

jwbonner, 2023-08-28:
The 2024 version of AdvantageScope will use a new format for custom assets, but it's still under active development. The online documentation linked in the repo is for the current development version. For adding custom models, you should reference the documentation built-in to the latest stable release (v2.3.0) by clicking the 📖 icon. The older docs are also online [here](https://github.com/Mechanical-Advantage/AdvantageScope/blob/v2.3.0/docs/CUSTOM-CONFIG.md).

With the 2023 format, everything is under "frcData" with no subfolders. See the older built-in frcData for reference: https://github.com/Mechanical-Advantage/AdvantageScope/tree/v2.3.0/frcData.

Once the 2024 betas are ready, all of the assets in the 2023 format will be upgraded automatically so don't worry about developing with the 2023 format right now.

mvog2501, 2023-08-30:
Thank you! This was able to resolve my issue. The problem was that the online documentation was for a more recent version of AdvantageScope than the most recent release. This could be resolved temporarily by adding a disclaimer at the top of the "Custom Assets" page. I believe I was also testing using assets created for the current version on `main` rather than the version of AdvantageScope that I was using.

Thank you so much for your help.

closed 2023-08-30 by mvog2501

jwbonner, 2023-08-30:
Disclaimers added! 👍

Thanks for the suggestion.