Mechanical-Advantage/AdvantageScope #94: Support comparing multiple log files at once

jlmcmchl, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
I'm trying to address a need in FRC where there isn't a good way to currently compare data across wpilogs in an easy manner. Currently, if I need to compare data in two wpilogs, I need two instances of AdvantageScope running, or otherwise use a custom tool.

**Describe the solution you'd like**
AdvantageScope should support loading one or more log files, where the "root" of each wpilog are sibling nodes, like:
- /first-wpilog/DriverStation/Enabled
- /second-wpilog/DriverStation/Enabled

Additionally, in order to make this work, support for synchronized timestamps is required. Custom offsets might be nice, but really here I'm looking to synchronize logs on common boolean topics like /DriverStation/Enabled, etc.

In a 'multi-log' mode, I'd also like the root of the topic "tree" to be a label on plotted data rather than a part of the topic name when rendered. If I want to look at a topic in one log, I want to look at the same topic in the other log. 

**Describe alternatives you've considered**
A possible alternative to this is to pre-process multiple wpilogs into a single input wpilog where all topics are prefixed by some identifying information. This leaves me with a lot of extra work by hand to compare data though, since I'd still need to drag and drop topics into plots every time I open advantage scope. I also can't synchronize timestamps without significant additional pre-processing.

If I run two separate instances of AdvantageKit, I can synchronize timestamps, but then I have to deal with 2x windows for everything.

**Additional context**
These details are not meant to be a complete image of what I envision this would become, but I hope this gives you an idea of what I'm looking to accomplish.

jwbonner, 2023-10-15:
One version of this is already implemented and will be part of the 2024 betas. If you merge a log with conflicting fields, all of its fields will be prefixed with "/MergedLog0/..." and count up as you add more files (the field names from the original log are unaffected). Merging is automatic based on the first enable, which is supported for several logging formats. As before, if logs are merged with no overlap in fields names (like a wpilog and DS log) there will be no prefixing.

This approach deals with each log as a separate set of prefixed fields, so you can use a subset of them for any visualization. It sounds like you would prefer that the fields from all of the logs are automatically displayed together, if I'm understanding correctly. I'm sure not which is the more intuitive behavior. My thinking was that the user would often want to compare data between only a subset of the merged logs, or show logs using different object types (like a robot as reference and ghosts from other logs). At a minimum I'd want to allow for that flexibility by some method, and I'm not sure how it would work if each field automatically represents itself across all merged logs. Let me know if you have suggestions about that.

jlmcmchl, 2023-10-18:
This gets me part of the way there - I think I'd prefer to select multiple logs in the `Open Log` dialog at which point a merge routine would occur, where each topic in both logs are prefixed / suffixed with an identifier. The option to suffix each topic would give me the UX I'm looking for in easily being able to compare equivalent topics from two logs together by grouping them.

I'd also like to be able to render metadata from two merged AdvantageKit logs on the metadata page. Is that something that could be done?

jwbonner, 2023-10-19:
I'll take a look at this soon. I like the idea of an optional suffix instead of a prefix, and I think the "non-specific affix" matches the current implementation (this is the reason for the generic "MergedLog0" prefix instead of a filename). Supporting AdvantageKit metadata shouldn't be difficult either, that's just an oversight.

With the idea of a multi-selector for merging, I think that's a good idea but it will require sensible limits on the number of logs/fields. Since AdvantageScope needs to cache all of the data for the visualizations to work smoothly, merging a large number of logs could quickly become unwieldy. The single-file selector discourages this inherently, but actual limits are probably necessary so users don't try to merge dozens of large log files together.

jwbonner, 2023-11-20:
This is partially addressed by a9f5b255ee040c05b4a978086360d7a86f8f19a1. Several log files can be opened together and merged ("Open Multiple..." replaces "Merge Log..."), and metadata is supported correctly. It's limited to 10 log files right now since caching more data than that tends to work poorly; analyzing long term trends isn't what AdvantageScope is designed for anyway, which would be the primary use case. It also doesn't support key suffixes yet, only prefixes. Suffixes get messy when dealing with arrays, structured data, mechanisms, etc so it'll require some significant effort to implement. I may look into that in the future, but for now the search function is the best option for jumping to the same field across different logs.