Mechanical-Advantage/AdvantageScope #102: Add boolean panel

lidor51, 2023-10-26:
The goal is being able to show boolean state of a variable in a noticeable way.
The suggestion it to add a "boolean tab", in which you can drag boolean variables, and they'll show as a LED (on/off) with the var name.

A rectangular green led would be sufficient for simplicity, and efficient arrangement when you drag in multiple booleans.

jwbonner, 2023-10-26:
You can currently drag out a series of boolean options to the line graph or table, which are also capable of showing the full history of the values instead of a single sample. What is the use case for this feature not covered by those options? (Keeping in mind that AdvantageScope is not a driver dashboard).

lidor51, 2023-10-30:
Some stuff are just much easier to debug/analyze when they're clearly visible and pops out.

jwbonner, 2023-12-01:
For the next release, many values (including booleans) will be displayed in the sidebar as shown below. Does this cover your intended use case? I think a more customizable indicator interface is probably out of scope, and better suited for other tools like Shuffleboard, Elastic, or FRC Web Components. Of course, the line graph/table views are also available for deeper analysis.

[image]

lidor51, 2023-12-01:
That's cute thanks for sharing, I think the minimum effort that will cover the intended use case is to put this same indicator for booleans also in the table interface.

jwbonner, 2023-12-29:
[v3.0.0-beta-7](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v3.0.0-beta-7) adds color indicators to booleans in the table interface, which appears to cover this use case.

closed 2023-12-29 by jwbonner