Mechanical-Advantage/AdvantageScope #99: Sidebar text does not render correctly

bznerd, 2023-10-26:
**Describe the bug**
The NT tables and sources have some kind of visual bug in the text rendering. Looks like it may be multiple layers of text.

**To Reproduce**
Happens with any NT connection.

**Screenshots**
[image]

**Version (required):**
Version: 3.0.0-beta-3
Distributor: FRC 6328
Platform: linux-x64
Build Date: 10/20/2023, 19:45:59 UTC
Electron 26.2.1
Chromium: 116.0.5845.188
Node: 18.16.1

**Additional context**
I'm not aware of any changes or significant updates to my desktop environment (wayland) recently, and have not experienced this before with the 2023 version of AdvantageScope.
I'm happy to install an older release to test if that would be helpful.

jwbonner, 2023-10-27:
Based on #77, it seems like Wayland and Electron haven't been cooperating very well. I don't know for sure if that's the root cause, but it's a strong possibility. I'll try to recreate it on my end with a Wayland desktop; it's possible I'll just need to switch Electron versions to fix it.

bznerd, 2024-01-23:
Any update on this or #77? I'm on the latest version now and still the same issue. If there's anything I can do to help diagnose and fix the issue let me know.
[image]

jwbonner, 2024-01-24:
I haven't been able to recreate this thus far, though I still suspect it's related to Chromium rendering on Wayland. I was experimenting with a Chromium flag for #77 which may be applicable here, and I believe Electron and Chromium have been improving their Wayland support. It would be helpful if you could try running one of the dev builds [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/runs/7633600754), which use the Wayland Chromium flag and the latest version of Electron/Chromium. I'll also dig more into trying to recreate this locally.

bznerd, 2024-01-26:
Figured out the source of it. Chromium doesn't play nice with the linux i915 intel gpu driver. Found reports of the issue popping up periodically and being patched only to come back again later. I'm not sure there's anything that can be done about in this project. I'll try that build and might try upgrading to Ubuntu 23.10 but I don't have a whole ton of hope.

bznerd, 2024-01-26:
Just tested the dev build and still same problem. The only workaround I can find is to run the app with --disable-gpu which seems to run fine with the exception of the 3d field view being basically unusable.

jwbonner, 2024-01-26:
Wow, thanks for tracking that down. If it's helpful, I could probably add a hidden preference to disable the GPU so you don't need the CLI option. Otherwise it sounds like there's not much else to be done on our side.

bznerd, 2024-01-26:
Nah it doesn't bother me. I'm going to close this issue but if I can find the motivation to do some research about chromium and the Mesa library or to test some old versions of AdvantageScope I might try and isolate the exact change that broke stuff and open an issue with chromium.

closed 2024-01-26 by bznerd

bznerd, 2024-03-14:
Looks like an intel driver update or something fixed this properly.