Mechanical-Advantage/AdvantageScope #104: Screenshot not shown in embedded docs

sciencewhiz, 2023-11-04:
**Describe the bug**
In AdvantageScope docs on Windows from WPILib build (beta 4), the example screenshot doesn't show

**To Reproduce**
Steps to reproduce the behavior:

1. Click on 📖

**Expected behavior**
Example AdvantageScope screenshot should be shown

**Screenshots**
[image]

**Version (required):**
[image]

sciencewhiz, 2023-11-04:
After loading AdvantageScope a second time, the screeenshot loaded

jwbonner, 2023-11-04:
This should be fixed now. It looks like there was an edge case on startup where the renderer didn't always receive the platform information before trying to set up image URLs.

commit 398a54b4 (2023-11-04): Make docs image URL fix platform agnostic (fixes #104)
closed 2023-11-04 by jwbonner