Mechanical-Advantage/AdvantageScope #110: Add UI to view DataLogEntry's metadata

Braykoff, 2023-11-30:
WPILib's DataLogEntry objects have a Metadata property, which allows a String (usually JSON formatted) to be set as an entry's metadata. This can be set through the constructor or the [setMetadata(String metadata)](https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/util/datalog/DataLogEntry.html#setMetadata(java.lang.String)) method. Currently, (to my knowledge), there is no way to view this data in AdvantageScope. WPI's DataLogTool has a column for this, and this is currently the only way to view it.

**Describe the solution you'd like**
I would like a tab, or other menu, that allows users to view the String contents of the metadata field of log data. Some of my logged data has additional details (such as unit) that would make it easier to go through logs.

katzuv, 2023-12-04:
Another small enhancement is to syntax highlight the text if it's valid JSON.

jwbonner, 2023-12-21:
Hopefully this addresses your use case. The metadata will now be shown when hovering on the field in the sidebar, and if the metadata is valid JSON it will be displayed in this key-value format.

[image]

commit 6edb70ff (2023-12-21): Improve wpilog metadata handling (closes #110)

Show when hovering on key in sidebar and include in wpilog export
closed 2023-12-21 by jwbonner