Mechanical-Advantage/AdvantageScope #114: Make unit conversions persistent

katzuv, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
Currently after setting a unit conversion for a field won't persist after the program relaunches.

**Describe the solution you'd like**
Make unit conversions, per key, persist after AdvantageScope is closed.

jwbonner, 2023-12-22:
The unit conversions on the line graph should already persist between restarts. Is there a case where you’re seeing that fail?

katzuv, 2023-12-22:
Unless I'm doing something wrong, it doesn't work. This is after a simple factor of zero:
[image]
After restarting, there is no factor set:
[image]
This doesn't work when using the from/to unit fields, too.

jwbonner, 2023-12-22:
Does the set of fields restore correctly? It would be helpful if, after setting some values, you can send the state JSON file. You can find it by going to "Help" > "Show Assets Folder" and then navigating up one level. The file is named "state-VERSION.json"

katzuv, 2023-12-22:
Here's the content. I'll add that the fields themselves aren't retained, if they should be.
```json
{
  "x": -8,
  "y": -8,
  "width": 1936,
  "height": 1056,
  "rendererState": {
    "sidebar": {
      "width": 300,
      "expanded": [
        "/AdvantageKit",
        "/AdvantageKit/RealOutputs",
        "/AdvantageKit/RealOutputs/Odometry",
        "/AdvantageKit/RealOutputs/Odometry/Robot"f
      ]
    },
    "tabs": {
      "selected": 1,
      "tabs": [
        {
          "type": 1,
          "legends": {
            "left": {
              "lockedRange": null,
              "unitConversion": {
                "type": null,
                "factor": 0
              },
              "fields": [
                {
                  "key": "NT:/AdvantageKit/RealOutputs/Odometry/Robot/translation/x",
                  "color": "#2b66a2",
                  "show": true
                },
                {
                  "key": "NT:/AdvantageKit/RealOutputs/Odometry/Robot/translation/y",
                  "color": "#e5b31b",
                  "show": true
                }
              ]
            },
            "discrete": {
              "fields": []
            },
            "right": {
              "lockedRange": null,
              "unitConversion": {
                "type": null,
                "factor": 1
              },
              "fields": []
            }
          },
          "title": "Line Graph"
        }
      ]
    }
  }
}
```
BTW this is my version info:
[image]

jwbonner, 2023-12-22:
The fields are supposed to be restored, along with the rest of the application state. Can you try running AdvantageScope in the console? You should be able to launch it by running this line in command prompt:

```
C:\Users\USERNAME\AppData\Local\Programs\advantagescope\AdvantageScope.exe
```

Does anything print out in the console while launching?

katzuv, 2023-12-22:
Nothing prints out. However, after upgrading to `v3.0.0-beta-6`. the issue goes away, and fields and conversions are restored. Not sure what was wrong but closing this. Thank you!

closed 2023-12-22 by katzuv