Mechanical-Advantage/AdvantageScope #115: A way to see events

katzuv, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
It's hard sometimes to understand when important events happened. For example, the motor's current got to zero because it hit a limit switch, but there's no easy way to see this and correlate between these two events.

**Describe the solution you'd like**
I was imagining something like a logging function in AdvantageKit that receives a logging message, severity level, and an optional JSON metadata field (could be a `Map` object) used for filtering or custom purposes (log type like "command started", "sensors" type; subsystem name, etc). Then you can see the events as you hover over the line graph or as lines inserted in the table view. Regarding the graph, I thought of something like in this [screenshot](https://github.com/Mechanical-Advantage/AdvantageScope/raw/main/docs/resources/line-graph/line-graph-1.png) -- events would be shown as dashed lines, similar to the timestamp, and when hovered would show the full message. A window to filter based on the aforementioned JSON will also be available.  
Something else that could be useful is attaching logs to subtables, so only when values from the relevant suitable are used will the associated events be shown.

jwbonner, 2023-12-22:
For the example you gave, you can log the value of the limit switch as a boolean on the line graph and show it as a discrete field. In general, events and state changes can be expressed as discrete fields (booleans, strings, etc). Does this cover your use case?

katzuv, 2023-12-22:
Can strings be shown on the graph?

jwbonner, 2023-12-22:
Yes, you can drag then to "discrete fields"

katzuv, 2023-12-22:
Oh, that's nice! Didn't know about that. That seems as a good solution for what I was looking for, thanks.
The filtering feature might still be useful.

jwbonner, 2023-12-22:
I'm not sure what you mean by "filtering." Fields are already organized by table, and you can easily toggle the visibility of each field on the graph.

katzuv, 2023-12-22:
For example, a team logs every second some string with state for each subsystem, but it's being logged to the same key. The team can add `{"subsystem": "elevator"}`/`{"subsystem": "drivetrain"}`, etc. to the string metadata, then filter only "elevator" state logs.

jwbonner, 2023-12-22:
In that case, the states of different subsystems should probably be logged to different keys. That could also keep them organized into tables.

katzuv, 2023-12-22:
That sounds reasonable. Thanks!

closed 2023-12-22 by jwbonner