Mechanical-Advantage/AdvantageScope #117: Enable keyboard navigation in keys search results and abbreviation search

katzuv, 2023-12-22:
It'd be nice to be able to navigate the keys search results menu with up/down keys. Also, abbreviation search might be useful. Not sure if that is the technical term, but that's what I mean:
[image]
It can also be more reserved as IntelliJ IDEA does that -- only prefixes of words are accepted (hence the "e" at the end will make no results shown).

jwbonner, 2023-12-22:
Currently, the search looks for any key where the search text appears. Is this different than your suggestion to match the IntelliJ behavior?

katzuv, 2023-12-22:
Yes, for example for the field "Auto Chocies", IntelliJ IDEA will match `ac` `auch` too (`Au`to `Ch`oices).

12Jeef, 2024-01-01:
I suggest [FuseJS](https://www.fusejs.io/) for abbreviation search.

jwbonner, 2024-02-07:
Both of these features are now implemented. You can use the keyboard (up, down, enter) to navigate through the search results, and it uses FuseJS as the backend so the search term doesn't need to appear identically in the key.

closed 2024-02-07 by jwbonner