Mechanical-Advantage/AdvantageScope #118: Previewed strings should fill available space

katzuv, 2023-12-22:
Strings are truncated even if there's plenty of space:
[image]

jwbonner, 2023-12-22:
This is intentional to avoid cluttering the sidebar, since filling it with too much text makes it harder to view and navigate at a glance. I think longer strings are best viewed with the table, console, or line graph (especially with the example of AdvantageKit's console logging).

katzuv, 2023-12-22:
I see. Perhaps then the shown length should be in some range depending on the width? So it will be 8 at the minimum but say 15 top if the sidebar is wide. But this is something that is down to personal preference too.

jwbonner, 2023-12-22:
Since the purpose of the sidebar is just to preview the value, I'm not sure there's a compelling reason to show more of the text. Thanks for the suggestion though.

closed 2023-12-22 by jwbonner

jwbonner, 2024-02-02:
Having gotten this suggestion again with #138, I experimented with changing the value length based on the sidebar width and I think it works reasonably well without causing too much clutter. Thanks for making the suggestion originally.