Mechanical-Advantage/AdvantageScope #120: Failed to Process Video (worked on previous version of AS)

jasondaming, 2024-01-03:
Tried to add [this](https://drive.google.com/file/d/18aJU8_mnFFuINp7vF5pmjjP8ompVhF0c/view?usp=sharing) .mov file to a video tab and got a "Failed to process video" error.

Beta 7 version

jasondaming, 2024-01-03:
I also uploaded the video to youtube and there was a problem processing that too!

jwbonner, 2024-01-03:
For AdvantageScope, it looks like this is because of an edge case in the ffmpeg parsing to determine the video dimensions (this was introduced in v3.0.0-beta-5 to support automatic match synchronization). With some improved logic, it seems to be fixed.

I'm not sure why YouTube would be having issues though. Could it be related to the fact that this is an HDR video?

closed 2024-01-03 by jwbonner

hschilling, 2024-08-01:
I am having this same problem. None of the3  methods for using the video work for me. I have the latest AdvantageScope and have tried both on a Mac and Windows laptops. 

For example, if I try to add https://www.youtube.com/watch?v=b34nDQ5cUyM I get the Error Failed to process video. There was a problem while processing the videos. Please try again.

commit d8d4a8b9 (2024-01-03): Improve dimension parsing for videos (fixes #120)