Mechanical-Advantage/AdvantageScope #123: Allow the WPILib Installer to update AdvantageKit on macOS

ThadHouse, 2024-01-09:
Changes in macOS Ventura make it so updating apps from other apps is blocked by default. AdvantageKit is the first app WPILib has installed, and its going to have trouble updating in the next release due to this. The solution is to specify in the AdvantageKit plist that it's allowed to be updated by the WPILib Team ID.

https://developer.apple.com/documentation/bundleresources/information_property_list/nsupdatesecuritypolicy

closed 2024-01-09 by jwbonner

ThadHouse, 2024-01-09:
That fix isn't going to work @jwbonner. We're not marked as an installer, so it needs to be using AllowProcesses

jwbonner, 2024-01-09:
This is using the WPILib team ID. The format for `AllowedProcesses` appears to be correct now based on the examples in the Apple article.

commit 34f2eb5e (2024-01-09): Add WPILib team ID as allowed updater for macOS (closes #123)