Mechanical-Advantage/AdvantageScope #124: Underscores in NT keys are interpreted as slashes in tree view

shueja, 2024-01-11:
**Describe the bug**
In the 2024.1.1 wpilib-packaged AdvantageScope version, `_` in an NT key is interpreted as a path separator.

**To Reproduce**
Steps to reproduce the behavior:

1. In OutlineViewer, start it in server mode and add `m_test/value` (type doesn't matter)
2. Connect to the local server from AdvantageScope
3. See the tree shown as `m/test/value`

**Expected behavior**
NT Tree view should allow underscores in keys without interpreting them as path separators.

**Screenshots**
[image]
[image]

With the reproduction steps above:
[image]
[image]

**Version (required):**
(The text in "About Advantagescope" isn't selectable or copyable)
[image]

shueja, 2024-01-11:
As the Monologue author, I can also add a flag that strips the `m_` and `k_` out of field names, but that's more of a convenience thing for Monologue users specifically.

jwbonner, 2024-01-11:
Adding exceptions for `m_` and `k_` on the AdvantageScope side seems reasonable, since I think it's pretty unambiguous in an FRC context that those are just prefixes. Obviously if you think it's useful to be able to strip them out in Monologue you could add that too.

caburum, 2024-01-12:
We use underscores in our variable names for units (like `wheelVelocity_mps`) and this has caused all of our logged variables to be harder to access. If lacking a convention in other log formats is a problem, perhaps only use slashes for the AdvantageKit and SmartDashboard/Shuffleboard sections.

jwbonner, 2024-01-12:
It looks like the only real example that doesn't follow the convention of `/` and `:` for separators is CTRE's signal names. In particular, the current approach may be problematic for Python teams that stick with snake case for field names. While it's frustrating to need special cases like this, it seems like the safer way forward is to handle CTRE's signals separately and default to not separating on underscores.

jwbonner, 2024-01-15:
That update should address all of the use cases correctly. By default, keys are only separated on `/` and `:`. Anything from a Phoenix source (where the key starts with "Phoenix6") will be separated on `/`, `:`, and `_`. Thanks for the feedback everyone.

commit a51e0e12 (2024-01-15): Remove underscore separator, except for Phoenix sources

Closes #124
closed 2024-01-15 by jwbonner