Mechanical-Advantage/AdvantageScope #125: "About Advantagescope" window does not have selectable/copyable text

shueja, 2024-01-11:
**Describe the bug**
See title.

**To Reproduce**
Steps to reproduce the behavior:

1. Open AdvantageScope
2. Click on "Help->About AdvantageScope"
3. Try to copy the text

**Expected behavior**
Able to select and copy the text there. Either allow selection or add a button that copies the contents to clipboard (or both).

**Version (required):**
Copy the contents of the "About AdvantageScope" window and paste it here. On Windows/Linux, click "Help" > "About AdvantageScope". On macOS, click "AdvantageScope" > "About AdvantageScope" in the menu bar.

... this is why this is an issue.
[image]

jwbonner, 2024-01-11:
A button to copy it to the clipboard seems like a good solution. When writing the template I only tried it on macOS, which lets you copy text from native popup windows.

jwbonner, 2024-01-13:
There is now a support for copying the text to the clipboard, which should work on all operating systems. Thanks for the suggestion.

closed 2024-01-13 by jwbonner