Mechanical-Advantage/AdvantageScope #127: Sidebar not showing all network tables entries

blaine141, 2024-01-13:
**Describe the bug**
I can not get 3d poses to show up on the left when following the "3D Field" tutorial.

**To Reproduce**
Steps to reproduce the behavior:
Follow the instructions listed in 3D Field to publish Pose3d objects. Both the array and single Pose 3D. They will not show up on the sidebar to select

**Expected behavior**
The NetworkTables entries appear on the left

**Screenshots**
Network Tables entries
[image]

No corresponding sidebar entries.
[image]

**Version (required):**
[image]

**Additional context**
Maybe it is related to the struct type? I have a Field2d showing up correctly. I am also running from a simulator

jwbonner, 2024-01-13:
Are you publishing the fields without a leading `/`? In the latest released version a leading slash is required for NT entries, though I just pushed a fix for that (12afa312baba019aa4ca15484459dc5d3cc6c489). It's definitely misleading that the examples in the docs don't include that.

Let me know if that seems to explain the issue, otherwise I can do some more digging.

blaine141, 2024-01-13:
Dude!!!! It worked!!! I was reverse engineering your code and using the developer console debugger. I would have never found that. 

Is the leading slash a NT standard? If so I can make a PR for the docs.

closed 2024-01-13 by blaine141

jwbonner, 2024-01-13:
The leading slash isn't required by NT, so the fix I pushed should make it work properly without that. That should take care of the issue with the docs, but thanks for the offer.

blaine141, 2024-01-13:
BTW, this tool is amazing! This is the best-designed solution I have ever seen introduced into FRC. I really hope everyone starts using this tool so we can see some cool auton and less bugs!