Mechanical-Advantage/AdvantageScope #128: 3d Custom Robot Asset

brennonyoo, 2024-01-14:
Hi I had a question about inputting custom robot models. I seem to have followed all of the conventions described in the documentation, I added a model in .glb and the config.json all in the proper format but I do not see any option in the AdvantageScope 3d field menu. Is there something further I need to do? I heard some people mention code but I didn't see any mention of that in the documentation so I'm not sure.

jwbonner, 2024-01-14:
To confirm, did you place the files in the folder that opens when you click "Show Assets Folder" in the menu bar? The folder for the robot model should be named "Robot_???", so the overall file structure looks like this:
* `userAssets/README.txt`
* `userAssets/Robot_???/config.json`
* `userAssets/Robot_???/model.glb`

If all of that looks correct, it would be helpful if you could send the config JSON file so I can try to recreate the issue.

brennonyoo, 2024-01-14:
Yes, here it is. I was having some issues with the format so I took the kitbot config and just changed the name, perhaps that might be where the issue is. 

https://pastebin.com/z32Tue0M

jwbonner, 2024-01-15:
I don't see any issues with the config JSON, so I suspect the problem is somewhere else. Here's what the folder structure should look like with the GLB file:

[image]

Could you confirm that your folder structure looks like this?

I see the robot appear under the "Robot" dropdown in AdvantageScope as shown below. Is it accurate that you're seeing other robots in that list but not "Luna"?

[image]

jwbonner, 2024-02-08:
Did you find a solution to this issue?

jwbonner, 2024-02-17:
I assume that you found the issue with the asset and got it to work. If you're still having trouble feel free to reopen this issue.

closed 2024-02-17 by jwbonner