Mechanical-Advantage/AdvantageScope #129: Swerve Tab shows modules on different positions

AndromedaHelix, 2024-01-17:
**Describe the bug**
While trying to publish my SwerveModuleStates to show them in AdvantageScope, I've noticed that the placement of each module is different that the one on the phyisical robot. The module has the order FR, BR, BL, FL, and I've made sure to select that option in the arrangement option, yet none appears to work. I've made sure to check that each of the modules' state is published on the correct order: 0 for FR, 1 for BR, 2 for BL, 3 for FL, and in AdvantageScope they are shown in the correct order, yet this order does not translate to the swerve tab. 

**To Reproduce**
Steps to reproduce the behavior:

1. Publish the swerve module states. I've tried this:
`@AutoLogOutput(key = "SwerveStates/Measured")
  private SwerveModuleState[] getModuleStates() {
    SwerveModuleState[] states = new SwerveModuleState[4];
    for (int i = 0; i < 4; i++) {
      states[i] = modules[i].getState();
    }
    return states;
  }`
2. Open AdvantageScope
3. Add SwerveTab
4. Add the published states to the red/blue states section of the swerve tab.

**Expected behavior**
A correct representation of the placement of each module on the Swerve Tab. 

**Version (required):**
```
Version: 3.0.1
Distributor: WPILib
Platform: darwin-arm64
Build Date: 1/6/2024, 07:31:54 UTC
Electron: 26.2.1
Chromium: 116.0.5845.188
Node: 18.16.1

```
**Additional context**
This is important because we are updating our swerve to 2024, and we can't see if the code is working if the display is incorrect.

jwbonner, 2024-01-21:
Thanks for finding this, the fix will be in the next release. It looks like this has been an issue for a while, but no one noticed because it didn't affect the more commonly used arrangements.

closed 2024-01-21 by jwbonner

AndromedaHelix, 2024-01-21:
Glad I could help :)