Mechanical-Advantage/AdvantageScope #130: sysid-test-state-Drive file appears in AdvantageScope but not in SysID

russellmarkwalter, 2024-01-22:
This could very well be an issue with the SysID program. I used URCL to log my four drive motors during the four sysid tests, and at the end I was able to download the log file and open it in AdvantageScope. At this point, the sysid-test-state-Drive file appeared with all of the other URCL stuff I needed, and the graphs looked pretty good. The next step in the docs was to open it in the sysid program, but when I opened the log file, everything BUT the UCRL and sysid-test-state-drive files were there for me to load into sysID. 
[image]
Note how there is no UCRL file under NT, and no sysid-test-state-Drive file.

russellmarkwalter, 2024-01-22:
https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/REV-LOGGING.md
Here are the instructions I followed. I didn't do anything that said "advantagekit" users because I didn't know what that was. Do I need to do that?

jwbonner, 2024-01-23:
Did you export the log file from AdvantageScope and open the exported (not the original) log in SysId? You'll also need to make sure to check "Include Generated" in the AdvantageScope export options.

If it still doesn't work after following those steps, it would be helpful if you could upload both the original log file and the exported version. You may need to put them in a zip archive to upload them to GitHub.

russellmarkwalter, 2024-01-24:
I reran the tests and exported them again and I was able to get the file open in SysID with all of the UCRL and sys-id-state stuff. However, when I hit load, SysID crashed. I probably just need to redo the test. Thanks for the help, Mr. jwbonner.

jwbonner, 2024-01-24:
Glad you got it to open correctly. In regard to the crashing, make sure you're running the latest version of WPILib (2024.2.1) and AdvantageScope (v3.0.2). There was a bug in the kickoff versions that caused SysId to crash while opening logs from AdvantageScope.

russellmarkwalter, 2024-01-24:
Alrighty! Thanks again Mr. Bonner.

closed 2024-01-24 by russellmarkwalter