Mechanical-Advantage/AdvantageScope #131: # not encoded when loading file urls

MichaelStatt-S, 2024-01-23:
**Describe the bug**
This was discoved when our computer's user folders contained a # to label the computer number. No downloaded assets rendered. Looking at the console, the file wasn't found. This was because the # in the url wasn't replaced with %23

**To Reproduce**
Steps to reproduce the behavior:

1. create a new custom asset (ex Field3d_2024Field#2
2. try to render that asset (ex change field to field 2
3. object doesn't render
4. file not found error in console
5. error contains non-encoded # 

**Expected behavior**
The objects should render / the # should be encoded

**Version (required):**
[image]

jwbonner, 2024-01-24:
Thanks for reporting this. The update I just pushed appears to fix the issue from my testing. You can download the development builds [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/runs/7633357903) if you want to try it before the next release (note that CI only runs AppImage builds for Linux).

MichaelStatt-S, 2024-01-24:
Looks good to me! Thanks

closed 2024-01-24 by jwbonner