Mechanical-Advantage/AdvantageScope #132: 3D View on M1 Macbook Pro crashes

CrispyBacon1999, 2024-01-23:
**Describe the bug**
If I add a robot position to the 3D view, the software hangs for 15-30 seconds, then proceeds to crash. It seems like it's only the webview that crashes as well, since the window itself stays open.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to 3D Field
2. Connect to robot (or simulator)
3. Put a robot pose on the field
4. Software crashes

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
Trying this with the dev tools open causes it to crash, but then pause on a breakpoint. Looks like the renderer is using ~3GB of memory at the point of crash.

**Version (required):**
Version: 3.0.2
Distributor: FRC 6328
Platform: darwin-arm64
Build Date: 1/21/2024, 05:25:41 UTC
Electron: 26.2.1
Chromium: 116.0.5845.188
Node: 18.16.1

jwbonner, 2024-01-24:
Which field and robot model do you have selected? My best guess is that something is going wrong while loading the model(s), which ends up creating way more objects than expected. If you're using a custom robot model, it would be helpful if you can send a copy of the model and configuration (either here or via email to software@team6328.org if you can't share publicly).

CrispyBacon1999, 2024-01-31:
I was getting this on 2024 field, using BotBot returns. Interestingly, it doesn't do this anymore on the new version, so something might have fixed it. I'll let you know if I keep getting it.

closed 2024-01-31 by CrispyBacon1999