Mechanical-Advantage/AdvantageScope #137: Unable to have more than 1 component

vichik123, 2024-02-02:
My team has 5 components all together this year, and yet even 2 components just crash the app or something because it doesn't load. Using a single component works perfectly. Also, we tried quite a few combinations of components so I don't think the size of the CAD is the problem.

[Here](https://github.com/Galaxia5987/robot-2024/blob/feature/elevator-sim/important-files/advantage-scope/cad/Robot_UNKNOWN/config.json) is an example of us trying to use the 3d simulation for our 2 stage elevator. [Here](https://github.com/Galaxia5987/robot-2024/blob/feature/elevator-sim/important-files/advantage-scope/cad/Robot_UNKNOWN/README.md) is a file explaining our components. The first component is the intake so we added that to the elevator config as well.

Are we doing something wrong with the json config? If so, how come AdvantageScope doesn't return an error?

jwbonner, 2024-02-03:
It looks like the issue is with the filenames for your component models. Per the documentation, there should be an underscore between "model" and the index, so it would look like "model_0.glb" instead of "model0.glb". The next version will already improve the error handling for invalid JSON configurations (see #134), but I'll add a check for the filenames too so there can be a more useful message displayed.

vichik123, 2024-02-03:
Ah I see, thank you. Although you know, I'm pretty sure it worked for a single component named model0.glb as well.

vichik123, 2024-02-03:
Just not the elevator for some reason.