Mechanical-Advantage/AdvantageScope #139: Default log download/open path to last used path

mjansen4857, 2024-02-05:
**Is your feature request related to a problem? Please describe.**
We like to organize our logs by having them in different locations depending on what we were testing, so when we download or open a log file we have to navigate to the path they are stored in every time.

**Describe the solution you'd like**
Default the path of the file chooser for log downloads and log opening to the last used path, similarly to how the import/export layout chooser does.

**Additional context**
Platform: Windows x64

jwbonner, 2024-02-06:
Since the Driver Station can now download log files itself, the idea was to standardize on using the "Log Files" folder for storing files (at least on Windows, macOS and Linux default to the last used path). However, I think it would be reasonable to show "Log Files" the first time the dialogs are opened and then the last used path afterwards.

jwbonner, 2024-02-07:
It will now default to the "Log Files" folder the first time one of these dialogs is used, and default to the previous location afterwards.

closed 2024-02-07 by jwbonner