Mechanical-Advantage/AdvantageScope #140: Setting Default Component Pose with Config file

hopeshunta, 2024-02-05:
We are trying to load our custom CAD into AdvantageScope. We have the base model and one component.

The model and component both load into AdvantageScope, but if we change the zeroedRotations and/or zeroedPositions, the default zero position stays the same.

Here is our config file, and I'm confident the structure and naming of the files is correct since we are seeing the model show up.
[config.json](https://github.com/Mechanical-Advantage/AdvantageScope/files/14172793/config.json)

Its possible that I have misunderstood something, but it seems like the zeroRotation/zeroPosition of the component "Model_0.glb" should vary if we change the config file. Unfortunately it does not.

jwbonner, 2024-02-06:
Here's the relevant section of the docs for tuning the zeroed position:

> When no component poses are provided by the user in AdvantageScope, the component models will be positioned using the default robot rotations and position. When component poses are provided by the user, the "zeroed" rotations and position are instead applied to bring each component to the robot origin. The user's poses are then applied to move each component to the correct location on the robot.

The idea is that if the user doesn't provide any component poses (as would happen by default), the components will be positioned as they appear when they were exported from the CAD and split apart; the position and rotation of the main robot model is used, but the "zeroedPosition" and "zeroedRotations" are ignored. For tuning, I would recommend publishing a series of 3D poses at the origin (just `new Pose3d()`) which will cause the zeroed position to be used instead. Then you can tune it to bring the components to the origin.

dustinhans, 2024-02-06:
You rock. Thank you!

jwbonner, 2024-02-08:
It just added more detailed documentation for how this process is meant to work: https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/CUSTOM-ASSETS.md#setup-process. If you got the issue resolved, I'll plan to close this issue.

jwbonner, 2024-02-09:
This appears to be resolved. Feel free to reopen the issue if you have additional problems.

closed 2024-02-09 by jwbonner