Mechanical-Advantage/AdvantageScope #141: High CPU on MacOS (Intel)

a1cd, 2024-02-06:
**Describe the bug**
High CPU usage by the Electron renderer helper on Intel Mac.

**To Reproduce**
Steps to reproduce the behavior:

Connect to logs on WPILIB AdvantageScope. I use a MacBook Air 2020 with a 1.1 GHz Quad-Core Intel Core i5 without struggle for almost everything else.

**Expected behavior**
A clear and concise description of what you expected to happen.
My computer shouldn't struggle to perform other basic tasks while AdvantageScope is running. I expect AdvantageScope to run without crippling my computer.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (required):**
Copyright © 2021-2024 FRC 6328

Version: 3.0.1
Distributor: WPILib
Platform: darwin-x64
Build Date: 1/6/2024, 07:31:54 UTC
Electron: 26.2.1
Chromium: 116.0.5845.188
Node: 18.16.1

**Additional context**
Add any other context about the problem here.
High CPU from `AdvantageScope (Helper)`. When connected to logs there are two `AdvantageScope (Helper)` processes, one `AdvantageScope` process, and a process named `AdvantageScope (GPU)`. All `AdvantageScope (Helper)` use a consistent 90%-100% cpu when connected to a simulator or robot.

jwbonner, 2024-02-07:
It would be helpful if you could run a performance profile in order to narrow down the root cause. Could you download the build of AdvantageScope from the artifacts [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/runs/7809411689)? This version includes several performance optimizations added since 3.0.1 and uses a different version of Electron that works around a bug with performance data exports. Here's the process to follow:

1. Connect to the robot/simulator such that you're seeing this issue (if the new version doesn't seem to fix it).
2. Open the developer tools under "View" > "Toggle Developer Tools" in AdvantageScope.
3. Go to the "Performance" tab, and click the circle on the upper left to begin recording. After ~10 seconds, click "Stop".
4. Click the download button to the right of the recording button, then save the JSON file and upload it here.

I know this is a bit tedious, but there are unfortunately a lot of potential causes for an issue like this and I haven't been able to recreate it thus far. The performance profile will provide useful data about which tasks are causing the issue.

a1cd, 2024-02-10:
Yes, I will send this over once I find some time to do this. I can probably do this some time tomorrow.

a1cd, 2024-03-01:
Ok, I just got around to updating my WPILib. I'm sorry it took so long! It seems that it is no longer an issue.

a1cd, 2024-03-01:
If you still want a performance profile on the latest version I can make one.

closed 2024-03-01 by a1cd