Mechanical-Advantage/AdvantageScope #145: Graph Improvement Ideas

Gold872, 2024-02-20:
I was playing around with the graph, and I had some ideas on new features to add to it. While it works great in its current state, it doesn't have some of the customizability and flexibility that is seen in other analysis tools such as the DS log viewer. I have a few ideas on new features/improvements:

1. Ability to "select" a viewing range
One useful feature would be able to easily select a viewing range by just dragging. It can be a bit annoying to zoom in and scroll through a massive log to see a variable over a 10 second window

I think dragging over this part should allow users to select a viewing range
[image]

2. Ability to change line thickness/color
I'm not sure whether or not this exists, but I couldn't find a way of changing the color of a specific variable. Whether or not it already exists, I think there could also be options for viewing it as a set of points instead of a line, different colors/line thicknesses, and maybe even a way of showing it as a bar graph instead of a line.

3. Ability to show discrete field points only when the value changes/when a boolean value is true
The primary use case of this is seeing when the robot is browned out/if we're connected. It was very useful in the DS log viewer and I would love to be able to use it in AdvantageScope.

cole-wilson, 2024-02-27:
I'd also like to add my support to this: I think being able to zoom on the line graph would be a very useful feature.

jwbonner, 2024-02-27:
@cole-wilson To be clear, you can already zoom in on the line graph by hovering the cursor over the graph and scrolling (up/down to zoom and left/right to pan). I believe the suggestion is to allow you to jump to a range directly by dragging on the timeline, which would also be a useful enhancement.

cole-wilson, 2024-02-27:
Oh okay, thank you! I hadn't discovered that yet.

closed 2024-09-14 by jwbonner