Mechanical-Advantage/AdvantageScope #155: Check for SymLink in userAssets

srimanachanta, 2024-03-13:
**Is your feature request related to a problem? Please describe.**
Most teams using ascope tend to keep a copy of their custom robot config and CAD models within their robot code directories in a folder, say "ascope_assets". It would be a helpful feature if I could just make a SymLink between the robot folder within the robot code repo and Ascope's userAssets folder. I've tried this and it seems not to work indicating that either I'm doing something wrong or Ascope doesn't follow SymLinks.

**Describe the solution you'd like**
It would be ideal if ascope would follow sym links placed into the userAssets folder.

**Describe alternatives you've considered**
Copy and pasting assets every time, this works but can be annoying as it requires two sets of configs to be changed every time.

jwbonner, 2024-03-13:
You can click "Use Custom Assets Folder" in the help menu to select a different folder for custom assets, which is intended for exactly this use case of storing assets alongside robot code.

jwbonner, 2024-09-14:
I seems like this use case is better addressed by the ability to set a custom assets folder.

closed 2024-09-14 by jwbonner