Mechanical-Advantage/AdvantageScope #157: 3d Field Not Rendering on Arch Linux.

person4268, 2024-03-14:
**Describe the bug**
Since some advantagescope version that I unfortunately do not remember, the 3d field does not render properly, and is just blank. The flatpak version of advantagescope I can verify works perfectly fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the 3d field.
2. Observe nonfunctionality

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Version (required):**
```
Version: 3.2.1
Distributor: FRC 6328
Platform: linux-x64
Build Date: 3/13/2024, 22:18:14 UTC
Electron: 26.2.1
Chromium: 116.0.5845.188
Node: 18.16.1
```

**Additional context**

goatfanboi23, 2024-03-28:
This is a known problem with Nvidia GPUs (on Wayland) and electron based apps. A workaround for now is to launch advantage scope with the "--use-gl=desktop" flag. This will most likely result in lag because it uses the CPU however a laggy field is better than no field.

person4268, 2024-12-01:
This isn't an issue for me anymore as of 4.0.0-beta-2

closed 2024-12-01 by person4268