Mechanical-Advantage/AdvantageScope #167: Download logs feature fails to connect with default IP of 10.00.AM.2

ArchdukeTim, 2024-04-18:
Had trouble downloading logs when connected to the Rio via Ethernet. I could SSH in, and advantagescope was showing the live robot view, but could not see log files. 

**To Reproduce**
- Set team radio to a 2-digit team number (e.g 78)

- Set advantage scope IP address to 10.00.AM.2 (e.g 10.0.78.2)

- Select "Download Logs"

- get unknown address error

- change robot ip in preferences to 10.0.AM.2 (e.g 10.0.78.2)

**Expected behavior**
- connects to robot and downloads logs

**Version (required):**
Will update when I'm back at computer

**Additional context**
Add any other context about the problem here.

importly, 2024-06-25:
Same issue, Exact same reproduce step but with a 2 digit team number 86. 
**Version:**  3.2.1
Only solution I found was plugging into the roborio with USB cable or unplugging the USB Drive

ArchdukeTim, 2024-06-26:
@importly Did my solution of using `10.0.86.2` instead of `10.00.86.2` work?

importly, 2024-06-26:
@ArchdukeTim Yes, using 10.0.86.2 instead of 10.00.86.2 worked for me. I did not try that at first. I think the simple way to solve this for any of the 99 teams trying to download logs with a regex expression. Thanks for the solution.

commit 0dd912fc (2024-09-14): Remove leading zeros from SSH address (fixes #167)
closed 2024-09-14 by jwbonner