Mechanical-Advantage/AdvantageScope #171: Add Support for Game Piece Poses

scarmain, 2024-07-18:
**Is your feature request related to a problem? Please describe.**
Feature suggestion for simulation purposes (could also be for physical testing), could we move the pieces shown in the 3d view by providing poses where they are, just like robot poses.  For some of the other visualizers, you can show both the robot/apriltag and a ghost, it would be cool if we can visualize game pieces in the 3d view tab.

**Describe the solution you'd like**
Options to add 3d poses for the game pieces.  That way we can make a simulation doing things like launching pieces and moving them for vision testing (with PhotonVision help).  Having the ghost options could show where the pieces physically were versus the vision code saw it.  There might need to be a second option/configuration for different game objects like cones/cubes or red/blue balls.

**Describe alternatives you've considered**
Originally, my idea was to use the WpiLib simulation to make the robot virtually appear on the field and create lessons based off the virtual robots, as getting physical robots are difficult.  With a Romi/XRP robot, it's hard to run mechanisms with them, as they aren't easily expandable (we were talking about making our own robots based off the Romi protocol).

I looked at XRC, but they don't allow custom robots or interfacing with your robot software.  I looked at Synthesis, it has like 80% of the features including robot code simulation and custom robots, but they have proprietary interfaces for intaking and shooting pieces, and cannot be interacted with through the simulation interfaces.

**Additional context**
I think the teams or a library should be responsible for the physics of the game pieces/walls/floors/collisions/etc.  This doesn't seem to be the responsibility of the visualizer.  Now maybe someone can contribute a basic physic engine, but I think that is out of scope for now.  Just visualizing where pieces are should be enough for now.

jwbonner, 2024-07-18:
Please see the documentation for 3D game piece objects: https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/tabs/3D-FIELD.md#game-piece-objects

scarmain, 2024-07-18:
I'm sorry, I'm dumb.  I was looking at the AdvantageScope I had (3.0.1, I think the initial release from WpiLib), and it didn't have notes then or game pieces from any older year.  When I look at a modern version (3.2.1), it is there.  Sorry :(

closed 2024-07-18 by scarmain