Mechanical-Advantage/AdvantageScope #172: Exporting a Log Crashes AdvantageScope on Linux (Mint 21.3)

error-27, 2024-07-25:
**Describe the bug**
When exporting a log in any format other than "CSV (List)", after saving the file in the file selection dialog, the app will freeze and then crash without creating the exported file.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to "File" > "Export Data"
2. Select any format except for "CSV (List)"
3. Click the save button
4. Save it anywhere in the file selector
5. See AdvantageScope freeze for a second before crashing
6. See that the exported log was not created

**Expected behavior**
I expect for the app to export a log and not crash.

**Screenshots**
None.

**Version (required):**
Version: 3.2.1
Distributor: FRC 6328
Platform: linux-x64
Build Date: 3/13/2024, 22:18:14 UTC
Electron: 26.2.1
Chromium: 116.0.5845.188
Node: 18.16.1

**Additional context**
Using the "CSV (List)" format successfully creates the log file and does not crash.

jwbonner, 2024-08-06:
I haven't been able to recreate this issue in my testing. Can you attach a log file where you see this occur? You may need to put it in a zip file to upload it to GitHub.

error-27, 2024-08-12:
Ok, I reproduced what i was doing that crashed it in the first place
However, it seems that even the exported CSV (List) is corrupted somehow, and AdvantageScope is refusing to open it
[export_crash_test.csv.zip](https://github.com/user-attachments/files/16589854/export_crash_test.csv.zip)

jwbonner, 2024-09-03:
We've been trying to recreate this on Linux with no luck. Have you seen this happen with multiple log files? It would be helpful if you can send us one of the original files you're exporting from.

error-27, 2024-09-08:
I'm exporting from the readings of a live simulation (I was doing SysID when I ran into this bug), and it does continue on repeat runs of the simulation. I just tested with an old Bot log from a competition and it exported fine, though.

So I'm thinking this might be an issue with either exporting the live feed, or possibly the use of AdvantageKit cycles, as the code I was working with used AdvantageKit while the previous logs were from a bot without AdvantageKit.