Mechanical-Advantage/AdvantageScope #173: Can't add videos

hschilling, 2024-08-03:
**Describe the bug**

This is the same issue as #120 .

None of the 3 methods for adding the match video work for me. I have the latest AdvantageScope and have tried both on a Mac and Windows laptops. No matter which method I try, I get the error shown in the Screenshots below. 

**To Reproduce**

1.  Load log files from a match
2. Click on the plus sign in the upper right and select Video
3. Any of these:

- Click the Blue Alliance icon
- after copying the URL for the match video, click the red clipboard icon
- download the match video and then click the gray document icon in AdvantageScope

See the error in the screenshot section

**Expected behavior**
Expected what is described on https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/tabs/VIDEO.md to happen. 

**Screenshots**
[image]

**Version (required):**
Version: 3.2.1
Distributor: FRC 6328
Platform: darwin-x64
Build Date: 3/13/2024, 22:18:14 UTC
Electron: 26.2.1
Chromium: 116.0.5845.188
Node: 18.16.1

**Additional context**
Add any other context about the problem here.

hschilling, 2024-08-04:
An update. I was able to make it work with a local file so I can do what I want to now. 

But the Blue Alliance method and the copy YouTube URL method still generate that same error.  Would be nice to have those work too but not super urgent any more.

jwbonner, 2024-08-06:
Thanks for reporting. It appears that this is due to a YouTube update that broke the version of ytdl AdvantageScope uses for YouTube downloads. Unfortunately this happens occasionally since YouTube regularly changes their APIs. I've updated the version of ytdl on `main` to the latest version, which fixes the issue.

In the meantime, you can either use a development build of AdvantageScope (artifacts are [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/workflows/build.yml)) or continue using local files.

closed 2024-08-06 by jwbonner

hschilling, 2024-08-10:
Thank you! 

The YouTube method now works.

But the TBA method doesn't for me. I don't get any errors. Just nothing happens. I checked my API key and it seems correct. I am trying this using your wpilog files from Einstein 5 in 2023.